pkwagner/yasdi2mqtt #1: Installationsfrage / Modbus => IP

AlexanderSch90, 2020-04-19:
Hallo Paul! Danke, dass du dieses Projekt geschrieben hast und geteilt hast. 
Wenn ich normalerweise Dockercontainer installiere, dann "pulle" ich das Image von Dockerhub. Mittels Watchtower bekomme ich so auch immer das neuste Image, wenn sich mal etwas ändern sollte (Bugfix etc.). Dein Image ist bei Dockerhub nicht vertreten. Hat das einen besonderen Grund? Wie kann ich das Projekt am einfachsten manuell installieren?  Mit einem `"docker build https://github.com/pkwagner/yasdi2mqtt"` bekomme ich es nicht hin.  

Dann noch eine inhaltliche Frage: Ich möchte einen Modbux => TCP-IP Adapter verwenden. Der Adapter ist aktuell unterwegs... Ist die Verwendung mit diesem Projekt möglich? [Link zum Adapter](https://de.aliexpress.com/item/32916950283.html?spm=a2g0x.search0302.3.2.2a024238EwpW1d&ws_ab_test=searchweb0_0,searchweb201602_0,searchweb201603_0,ppcSwitch_0&algo_pvid=553763ab-01a9-4f6b-8f70-e6eea1309e0b&algo_expid=553763ab-01a9-4f6b-8f70-e6eea1309e0b-0)

Hintergrund ist, dass ich ursprünglich SolarView verwenden wollte, da ich keine andere Anwendung gefunden hatte die Daten von den alten SMA-Wechselrichtern auszulesen. MQTT+InfluxDB+Grafana ist aber eigentlich mein Ziel, welches ich mit diesem Projekt vielleicht gefunden habe ;-) 
PS: SolarView nutzt diese Modbus zu TCP-IP Adapter. Siehe hier: [Installationsanleitung SMA-Proxy SolarView](http://www.solarview.info/SMAP_Installieren.pdf)

Vielen Dank für deine Antworten, ich denke diese helfen später auch anderen weiter. 

Viele Grüße

pkwagner, 2020-04-19:
Hey,

freut mich, dass Du Dich für das Projekt interessierst! Tatsächlich ist der Container derzeit noch nicht im Docker Hub, vielleicht hole ich das bei Gelegenheit mal nach. Wie Du richtig geschrieben hast, lässt sich das Projekt entweder aus einem lokalen Verzeichnis oder alternativ einfach per `docker build -t yasdi2mqtt https://github.com/pkwagner/yasdi2mqtt.git` bauen.

Anscheinend gibt es da Inkonsistenzen zwischen den verschiedenen Linux-Kerneln, bei mir kompiliert nach wie vor alles sauber. Mit dem Fix, den ich gerade comitted habe, sollte der Build aber nun auch bei Dir durchlaufen (einfach nochmal oben genannten Command eingeben).

Die Frage nach dem Modbus-Adapter kann ich zwar nicht aus eigener Erfahrung beantworten, prinzipiell sollte aber alles, was mit YASDI funktioniert, auch hier laufen. Die von Dir genannte `libyasdi_drv_ip.so` kann ich zwar leider allein aus Urheberrechtsgründen nicht direkt mit ins Projekt aufnehmen, mit dem `-v /pfad/zur/libyasdi_drv_ip.so:/etc/yasdi2mqtt/libyasdi_drv_ip.so` Flag hinter `docker run` solltest Du die Datei aber in den Container mounten können. Als Pfad in der `yasdi.ini` musst Du dann folglich `/etc/yasdi2mqtt/libyasdi_drv_ip.so` angeben.

Mit SolarView habe ich keine Erfahrungen, Dein angedachtes Setup mit InfluxDB und Grafana klingt aber sinnvoll und ist hier so auch im Einsatz. Zusätzlich dazu nutze ich Mosquitto als MQTT-Broker und Telegraf als MQTT-InfluxDB-Adapter.

Ich hoffe Dir damit erst einmal weitergeholfen zu haben.

AlexanderSch90, 2020-04-19:
Wow, danke für die ausführlichen Infos! Ich kann bestätigen, dass die Installation nun funktioniert. 

Zum Thema modbus => IP. Ich werde es testen sobald die Hardware da ist... Welche Hardware nutzt du denn eigentlich?

Der Docker run Befehl aus der Readme muss doch dann für mich um die beiden Punkte
```
--device /dev/ttyUSB0:/dev/ttyUSB0 \ 
   -v <project_dir>/devices:/etc/yasdi2mqtt/devices \ 
```
gekürtzt werden und mit dem `-v /pfad/zur/libyasdi_drv_ip.so:/etc/yasdi2mqtt/libyasdi_drv_ip.so`
ersetzt werden oder?

Beim Docker-run kommt der folgende Fehler: 
```
Alexander@ServerSchulte:~$ sudo docker run    -v /volume1/docker/yasdi2mqtt/libyasdi_drv_ip.so:/etc/yasdi2mqtt/libyasdi_drv_ip.so    -v /volume1/docker/yasdi2mqtt/yasdi.ini:/etc/yasdi2mqtt/yasdi.ini:ro    -e YASDI_CONFIG="/etc/yasdi2mqtt/yasdi.ini"    -e YASDI_DRIVER_ID="0"    -e YASDI_MAX_DEVICE_COUNT="1"    -e YASDI_UPDATE_INTERVAL="30"    -e MQTT_TOPIC_PREFIX="/solar/inverter"    -e MQTT_SERVER="10.4.4.13"    -e MQTT_PORT="1888"    yasdi2mqtt
10:59:17 INFO  src/main.c:61: Configuration | yasdi_config = /etc/yasdi2mqtt/yasdi.ini
10:59:17 INFO  src/main.c:62: Configuration | yasdi_driver_id = 0
10:59:17 INFO  src/main.c:63: Configuration | yasdi_max_device_count = 1
10:59:17 INFO  src/main.c:64: Configuration | yasdi_update_interval = 30
10:59:17 INFO  src/main.c:65: Configuration | mqtt_topic_prefix = /solar/inverter
10:59:17 INFO  src/main.c:66: Configuration | mqtt_server = 10.4.4.13
10:59:17 INFO  src/main.c:67: Configuration | mqtt_port = 1888
10:59:17 INFO  src/main.c:68: Configuration | mqtt_user = (null)
10:59:17 INFO  src/main.c:69: Configuration | mqtt_password = (null)
10:59:17 INFO  src/mqtt_client.c:72: Connection to mqtt broker established
10:59:17 FATAL src/yasdi_handler.c:42: Error while initializing yasdiMaster: -1
10:59:17 FATAL src/main.c:86: Unable to initialize yasdi_handler
```
Danke!

pkwagner, 2020-04-19:
Schön, dass das geklappt hat.

Die erste Zeile kannst Du in der Tat auslassen, `-v <project_dir>/devices:/etc/yasdi2mqtt/devices` würde ich aber beibehalten (wobei `<project_dir>/devices` einfach ein leeres Verzeichnis ist). YASDI nutzt das als Cache für die angeschlossenen Wechselrichter. Damit sparst du Dir ab dem zweiten Start die 1-2 Minuten, die YASDI initial braucht, um Gerätedaten herunterzuladen.

Ich habe hier einen Raspberry Pi mit RS485-USB-Adapter am Start, der sozusagen den gesamten Stack von yasdi2mqtt bis Grafana übernimmt. Der Fehler oben hängt vermutlich damit zusammen, dass YASDI noch keine Verbindung zum Adapter aufbauen kann.

Falls das später nicht klappt, kann es ganz interessant sein, dein Debug-Output von YASDI zu aktivieren. Dazu dann einfach das Projekt lokal herunterladen und in der `Dockerfile` den Parameter `YASDI_DEBUG_OUTPUT=0` durch `YASDI_DEBUG_OUTPUT=1` ersetzen. Danach muss dann nochmal mit `docker build -t yasdi2mqtt /pfad/zum/projekt` kompiliert werden.

pkwagner, 2020-04-19:
Was mir gerade noch aufgefallen ist: Vermutlich kannst Du Dir die `libyasdi_drv_ip.so` sogar sparen. YASDI sollte das per Default können, dazu einfach `Driver0=yasdi_drv_ip` setzen (vgl. [1]).

[1] https://github.com/konstantinblaesi/yasdi/blob/3a0ba932c023c100c2b6e020c775705205402815/samples/sample1/yasdi.ini

AlexanderSch90, 2020-04-19:
Top danke für die Info. Da war übrigens noch ein Fehler in der .ini meinerseits. Der Container startet nun, aber es kommen keine Werte per MQTT an, was ich erstmal als logisch empfinde, da ja auch nichts angeschlossen ist... (modbus => IP Hardware fehlt ja noch...) 
Ich melde mich wenn die Hardware da ist ob es klappt oder nicht und werde dann ggf. wenn es für dich OK ist auch "Werbung" hierfür machen. :-)

AlexanderSch90, 2020-05-04:
hey heute kam der mosbus => IP-Adapter. Einbindung klappt, wenn man dem Docker noch folgenden Befehl mitgibt: -p 24273:24273/udp \ => yasdi erstellt quasi einen UDP-Server welcher auf dem Port auf Eingang wartet. Vielleicht nimmst du das in die Readme auf.

Leider kommen jedoch noch keine Daten an. Ich denke aber ich bin zu 99% am Ziel. Hast du noch eine Idee. Anbei der Log aus Docker:
```
18:46:56 INFO  src/main.c:61: Configuration | yasdi_config = /etc/yasdi2mqtt/yasdi.ini

18:46:56 INFO  src/main.c:62: Configuration | yasdi_driver_id = 0

18:46:56 INFO  src/main.c:63: Configuration | yasdi_max_device_count = 1

18:46:56 INFO  src/main.c:64: Configuration | yasdi_update_interval = 30

18:46:56 INFO  src/main.c:65: Configuration | mqtt_topic_prefix = /solar/inverter

18:46:56 INFO  src/main.c:66: Configuration | mqtt_server = 10.4.4.13

18:46:56 INFO  src/main.c:67: Configuration | mqtt_port = 1888

18:46:56 INFO  src/main.c:68: Configuration | mqtt_user = (null)

18:46:56 INFO  src/main.c:69: Configuration | mqtt_password = (null)

18:46:56 INFO  src/mqtt_client.c:72: Connection to mqtt broker established

18:46:56 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...

18:46:56 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...

18:46:56 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...

18:46:58 DEBUG src/yasdi_handler.c:247: Event 0 fired for device 1

18:46:58 INFO  src/yasdi_handler.c:271: Device 1 went online

18:46:59 DEBUG src/yasdi_handler.c:247: Event 0 fired for device 161

18:46:59 INFO  src/yasdi_handler.c:271: Device 161 went online

18:46:59 ERROR src/yasdi_handler.c:277: Device count exceeds active_devices array size (size: 1)

18:46:59 DEBUG src/yasdi_handler.c:247: Event 0 fired for device 329

18:46:59 INFO  src/yasdi_handler.c:271: Device 329 went online

18:46:59 ERROR src/yasdi_handler.c:277: Device count exceeds active_devices array size (size: 1)

18:47:06 DEBUG src/yasdi_handler.c:247: Event 2 fired for device 3

18:47:06 DEBUG src/yasdi_handler.c:292: Device detection finished with 3 devices

18:47:26 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...

18:47:29 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 27 seconds...
```

pkwagner, 2020-05-04:
Sehr gut! Die Portfreigabe nehme ich gerne noch mit in die README auf.

Generell sieht es so aus, als hättest du drei Wechselrichter verbunden. Damit alle Geräte erfasst werden, muss der entsprechende Parameter `yasdi_max_device_count` also noch angepasst werden, sonst werden nur die Daten eines einzigen Gerätes abgerufen.

Wenn zwischen Value Fetch und Sleep keine Debug- bzw. Fehlermeldungen mehr kommen, sollte eigentlich alles geklappt haben. Schau daher vielleicht nochmal nach, ob Du den Subscribe wirklich auf den MQTT-Channel `solar/inverter/#` machst. Die Solardaten werden nämlich nicht direkt nach `solar/inverter`, sondern nach `solar/inverter/<sn>` geschrieben (`<sn>` ist dabei die jeweilige Seriennummer). Im Endeffekt sollten in Deinem Fall also drei verschiedene Channel genutzt werden.

AlexanderSch90, 2020-05-05:
Guten Morgen Paul! Wir haben 5 WR entsprechend habe ich den Wert auf 5 gesetzt. Nun läuft alles wie geschnitten Brot! Ich danke dir!

closed 2020-05-05 by AlexanderSch90