pkwagner/yasdi2mqtt #2: Zu viele MQTT-Meldungen bei 5WR

AlexanderSch90, 2020-05-05:
Hallo Paul,

kann es sein, dass zu viele MQTT Nachrichten versendet werden? Ich bekomme alle 30 Sekunden folgende Warnung auf meinem MQTT-Server:

```

mqtt.0 | 2020-05-05 18:49:16.392 | warn | (4791) Client [yasdi2mqtt] Message 6588 deleted after 11 retries
-- | -- | -- | --
mqtt.0 | 2020-05-05 18:49:12.389 | warn | (4791) Client [yasdi2mqtt] Message 6589 deleted after 11 retries
mqtt.0 | 2020-05-05 18:49:12.389 | warn | (4791) Client [yasdi2mqtt] Message 6590 deleted after 11 retries
mqtt.0 | 2020-05-05 18:49:10.483 | warn | (4791) Client [yasdi2mqtt] Message 6591 deleted after 11 retries
mqtt.0 | 2020-05-05 18:49:08.480 | warn | (4791) Client [yasdi2mqtt] Message 6592 deleted after 11 retries

```

pkwagner, 2020-05-05:
Falls Du ioBroker als MQTT-Broker benutzt, scheint das ein bekanntes Problem zu sein [1]. Ich selbst habe hier mosquitto im Einsatz, das macht da absolut gar keine Probleme.

Falls Du Deinen Broker nicht wechseln möchtest, hilft es wahrscheinlich auch, das MQTT QOS-Level herabzusetzen. Dazu musst Du in `src/mqtt_client.c` `#define MQTT_QOS 2` durch `#define MQTT_QOS 0` ersetzen und dann nochmal neu kompilieren. Dann wird aber auch nicht mehr garantiert, dass jede Nachricht auch wirklich (genau einmal) ankommt.

[1] https://forum.iobroker.net/topic/32143/mqtt-problem

AlexanderSch90, 2020-05-05:
Ja du hast recht. Es ist iobroker als MQTT-Broker. Ich bekomme jedoch trotz der warnings die Werte geliefert. Sehe es dann als Iobroker Problem => hier gelöst... 

Darf ich fragen wie du die Daten aufbereitest? 
Meine Idee war die Werte per mqtt an iobroker zu senden und dann das JSON-Objekt aufzuteilen in die einzelnen Attribute. Gibt es eine Legende was die einzelnen Werte aussagen? h-On, H-total, E-Total etc...
Angenommen ich habe es geschafft das JSON zu trennen: Die einzelnen Datenpunkte dann in einer Influx DB loggen und mit Grafana auszuwerten und bei Ausfall eines Inverters ein Alarm abzugeben. 
Vielleicht machst du es ja ähnlich (einfacher oder auch mit iobroker) und könntest es kurz beschreiben. 
Erstes Problem bei meiner Idee: Aus der JSON einzelne Datenpunkte machen. Hatte immer gehofft es kommen die einzelnen Datenpunkte an. Du hattest dich aber für eine JSON entschieden... 

Ich hoffe es stört dich nicht, dass ich dein Github Issue dafür missbrauche... Vielleicht hast du ja ansonsten Interesse per Telegram zu schreiben: Mein Nutzername ist @Alexander_Schulte

pkwagner, 2020-05-05:
Alles gut, vielleicht hilft unser Thread ja dem ein oder anderen weiter ;)

Um die Werte in die InfluxDB zu bekommen, führt eigentlich kein Weg an Telegraf vorbei. Da spielt es dann auch keine Rolle mehr, ob alle Werte gebündelt oder nach Kanälen aufgeteilt ankommen. Ich hatte mich damals gegen Letzteres entschieden, weil dabei dann doch recht viel Overhead entsteht. Meine Config für Telegraf findest Du [hier](https://termbin.com/yza0), der relevante Teil befindet sich in den Sektionen `outputs.influxdb` und `inputs.mqtt_consumer`.

Das Grafana-Setup ist dann ziemlich straight-forward, das läuft komplett über das Webinterface ab. Darüber kannst Du dann auch die Visualisierungen bzw. die Alarme einrichten, bei mir sieht das aktuell so[1] aus. Zusätzlich dazu habe ich noch ein paar `CONTINOUS QUERIES` direkt in InfluxDB eingerichtet, die die Daten sowohl 5-minütlich als auch täglich downsamplen.

Die einzelnen Werte musst Du Dir teilweise einfach ergoogeln, die Bedeutung ist meist in irgendwelchen SMA Service Manuals versteckt. Was ich bisher herausgefunden habe:

<details>
<summary>Legende für Parameter</summary>

```
{"sn":000,"time":1586608779,"values": {
   "Iac":12580, // Stromstärke AC (Stromnetz) [A]
   "Uac":239, // Spannung AC (Stromnetz) [V]
   "Fac":50.019998881965876, // Frequenz AC (Stromnetz) [Hz]
   "Pac":3006, // Leistung AC (Stromnetz) [W]
   "Zac":0.17200000816956162,
   "Riso":10000,
   "dI":4,
   "Upv-Ist":416, // Spannung DC (PV-Anlage) [V]
   "PPV":3013, // Leistung DC (PV-Anlage) [W]
   "E-Total":45358.538154414855, // Eingespeiste Leistung [kWh]
   "h-Total":47797.772832013434, // Betriebsdauer [h] (?)
   "h-On":51654.766385075491, // Betriebsdauer [h] (?)
   "Netz-Ein":9012,
   "Seriennummer":000,
   "E-Total DC":45694.108978657052,
   "Status":"Mpp",
   "Fehler":"-------"
}}
```
</details>

Den QOS-Wert kannst Du übrigens sowohl in yasdi2mqtt als auch in Telegraf ruhig auf 0 setzen, ich denke der Packet-Loss zwischen zwei Containern ist überschaubar.

---

<details>
<summary>[1] Screenshot Grafana</summary>

[image]
</details>

pkwagner, 2020-05-06:
Übrigens: Telegraf hat auch ein Output-Modul für MQTT. Damit sollte es problemlos möglich sein, den JSON-Payload in mehrere Nachrichten aufzusplitten und für andere Zwecke zu nutzen.

AlexanderSch90, 2020-05-10:
Hallo Paul, dein Support ist der Wahnsinn... 
Zunächst: Das MQTT QOS-Level herabzusetzen brachte die Lösung im iobroker. Für mich so OK.

Habe mich letztlich doch für die iobroker-Lösung entschieden und habe nun alles was ich haben wollte. Mein Weg also yasdi2mqtt => iobroker => Influxdb => Grafana. Somit musste ich mich nicht noch mit Telegraf auseinandersetzen und habe alle Infos in meinem Smarthome ohne zusätzliches System (zusätzliche Fehlerquelle). 
Mit dem passenden Javascript ist es ja ziemlich einfach möglich den JSON-Payload in einzelne Datenpunkte zu parsen. Anbei zur Info noch mein Grafana-Dashboard. Da das logging nun erst ein paar Tage alt ist, sieht es noch nicht so schön aus, aber man kann ganz gut erkennen wo es sich hinentwickelt. Ein bisschen Platz habe ich ja noch für weitere Infos. Mal schauen was ich draus machen... Alles in allem ein super Projekt und vielen Dank für deine Hilfe!

https://www.dropbox.com/s/cp5tzh8vp5nyg3p/PV-Grafana.png?dl=0

pkwagner, 2020-05-10:
Großartig, freut mich! Der Screenshot sieht wirklich top aus, vielleicht muss ich da bei Gelegenheit auch mal nachlegen ;)

AlexanderSch90, 2020-07-27:
Hey finde es cool, dass du nun auch im Docker-Hub bist. 
Falls du noch was optimieren willst: Eine ENV für das MQTT QOS-Level wäre cool, dann könnte man es auch als Iobroker-Nutzer Plug and Play nutzen. 
Alternativ würde ich vorschlagen das QOS-Level standardmäßig auf 0 zu setzen. Es handelt sich ja hier eh immer um "Fortgeschriebene Infos". Ob die jetzt nur "genau einmal" gesendet werden oder nicht ist ja bei diesem Anwendungszweig völlig egal... 

Viele Grüße

pkwagner, 2020-07-27:
Ich hab die Mittagspause gleich mal dazu genutzt, das QOS-Level ist jetzt als Umgebungsvariable drin 🚀 

Die Dokumentation dazu steht schon in der README, es dauert aber wohl noch so 15-20 Minuten, bis die Pipeline das dann auch auf Docker Hub aktualisiert hat.

closed 2020-07-27 by pkwagner