pkwagner/yasdi2mqtt #3: Integration for Arduino/ESP32

chrishrb, 2021-03-03:
Have you thought about an integration for Arduino/ESP32? Is this even possible?

pkwagner, 2021-03-03:
The underlying `yasdi` library requires a fully-functional operating system like Linux or Windows, so unfortunately, compiling the project for microcontrollers won't succeed. If you wanted to use an ESP for this, you needed to re-implement the whole (outdated) SMA protocol just to save a few bucks. 

However, `Modbus IP` might be another great option for you. These adapters[1] connect RS485 directly to a Wi-Fi network and usually cost even less than an Arduino. You'll still need to set up `yasdi2mqtt` somewhere on your network, but the server doesn't need to sit near the inverters, hence you can simply use your existing home server for it. Theoretically, you could even run `yasdi2mqtt` in the cloud then.

You'll also find some further reading regarding `Modbus IP` here[2]. Maybe @AlexanderSch90 can give you more advise, as he has this setup in use for his inverters.

[1] https://www.aliexpress.com/item/32916128353.html (just an example, didn't test them tho)
[2] https://github.com/pkwagner/yasdi2mqtt/issues/1

chrishrb, 2021-03-03:
I think that needs more work then it saves me money, so i will try this repository on my raspberry pi.
I wasn´t sure if the yasdi library would work with a microcontroller - but without this lib its not worth spending time. 

Thank you for the answer and this very nice repository!

closed 2021-03-03 by chrishrb

pkwagner, 2021-03-03:
Great choice, let me know if you encounter any problems.