pkwagner/yasdi2mqtt #4: Verkabelung von Raspberry mit WR

chrishrb, 2021-03-05:
Hallo, 
ich bins nochmal, jetzt mit ner anderen Frage (diesmal in deutsch, habe erst jetzt gesehen, dass du ja aus Deutschland bist): Wie hast du die WR mit dem raspberry verkabelt?
Irgendwie gibt es überall widersprüchliche Angaben, jetzt wollte ich da dich mal fragen. Ich habe mir einen Raspberry pi zero und einen entsprechenden Adapter gekauft: https://www.amazon.de/gp/product/B07B667STP/ref=ppx_yo_dt_b_asin_title_o01_s02?ie=UTF8&psc=1

Diesen muss ich ja einfach mit der seriellen Schnittstelle des Raspberrys verkabeln und dann noch den Adapter mit den WR, aber wie? Oder brauche ich den Adapter gar nicht? Bzw. wie hast du das ganze verkabelt? Ich habe zwei SunnyBoy 3000.

Danke im Voraus!
Grüße Christoph

pkwagner, 2021-03-05:
Ich habe mir damals diesen[1] USB-Adapter geholt und RX/TX/GND entsprechend mit den drei Kabeln aus den WR verbunden. Ich hatte dann direkt ein Device `/dev/ttyUSB0`, das `yasdi` so auch angenommen hat. Ob das mit Deinem Adapter auch klappt, kann ich natürlich nicht sagen.

Der erste Schritt wäre bei der vermutlich der gleiche (RX/TX heißen halt A+/A-). Statt per USB musst Du die linke Seite dann noch an den entsprechenden GPIO-Pins (RX/TX/GND/5V) mit Deinem Pi verbinden. Soweit ich weiß gibt es auch für die Pins direkt vorinstallierte Devices[2], über die Du dann mit den WR kommunizieren kannst.

Generell würde ich mit dem Zero ein bisschen aufpassen, meines Wissens basiert der auf der selben `arm/v6`-Architektur wie die erste Generation. `yasdi2mqtt` sollte kein Problem sein, aber mega viele andere Docker-Container sind dafür nicht ausgelegt, da musst Du dann manuell nachkompilieren. Ich bin damals deshalb auf einen 3A+ umgestiegen.

[1] https://www.amazon.de/-/en/Digitus-USB-to-Serial-Adapter/dp/B007VZY4CW
[2] https://www.raspberrypi.org/documentation/configuration/uart.md

chrishrb, 2021-03-05:
okay, also ich will auf dem raspberry nur yasdi2mqtt laufen lassen, deswegen kein problem. 
RX, TX ist bei den WR beschriftet oder? bzw. welche Pins sind das? und dann noch am letzten nen Jumper rein (als Abschlusswiderstand) oder? Diese Anleitung hat mich ehrlich gesagt, richtig verwirrt: 

https://files.sma.de/downloads/SB3000-11-EE3801.pdf (seite 55)

pkwagner, 2021-03-05:
Puh, ich hab damals glaube ich einfach alle 6 Möglichkeiten durchprobiert. Keine Ahnung wie gesund das für alle Beteiligten war. 

Widerstände habe ich keine drin, soweit ich weiß regelt das alles der RA485-Adapter. Ich bin aber auch kein Elektrotechniker, von daher kann ich da keine fundierte Aussage treffen. Stand jetzt (1 Jahr Betrieb etwa) hatte ich jedenfalls noch keinen einzigen Fehler.

pkwagner, 2021-03-05:
Gerade erst einen Blick in die PDF geworfen, vorausgesetzt das korrekte PiggyBack ist installiert, gibt Seite 52 dann nicht die korrekte Verkabelung vor?

chrishrb, 2021-03-05:
Ich will es über RS485 laufen lassen (also alle WR in einem Bus) also nein, nicht ganz. 
Aber ja, meine Bauteile kommen heute an, dann werde ichs einfach mal ausprobieren. Den Abschlusswiderstand hast du nicht gesetzt bei dir oder, wie auf Seite 55 angegeben? 
Ich kenne mich nur einigermaßen mit dem ganzen aus, vom Studium und meinem CAN-Gateway. Da ging bis jetzt auch immer alles gut :D

pkwagner, 2021-03-05:
Ach ja, das mit RS232 habe ich übersehen, bei mir ist auch RS485 mit 2 WR am Start.

Ja, bei mir sind die WR direkt mit dem Adapter verbunden. Allerdings ist ja ein Breakout-Board vorgeschaltet (siehe Amazon-Foto), keine Ahnung ob es das vielleicht schon macht.

konstantinblaesi, 2021-03-05:
Bei mir ist es
* Raspberry Pi
* [dieser RS485 zu USB adapter](https://www.amazon.de/dp/B00I9H5J02?ref_=pe_1365651_54849071)
* ein Piggy-Back pro Wechselrichter (eBay)
* [dieses Datenkabel](https://www.conrad.de/de/p/lapp-2170204-1-busleitung-unitronic-bus-2-x-2-x-0-22-mm-violett-meterware-600618.html)
* 4 Wechselrichter
  * 3x ein 3000er Modell
  * 1x ein 1100er Modell

Der Aufbau ist wie in den SMA Dokumenten gefordert RS485 daisy-chained. Ich habe auch so einige SMA Dokumente gelesen, in jedem habe ich etwas gefunden, dass mein Verständnis des Setups verbessert hat :D

RasPi -> USB zu RS485 Konverter -> Inverter 1 -> Inverter 2 -> Inverter 3 -> Inverter 4 

Inverter 4: Jumper zum Terminieren des RS485 BUS gesetzt

Ich habe absolut null Erfahrung mit seriellen Schnittstellen bzw. Eletronik und mein Bauchgefühl ist, dass es ein billigeres Datenkabel auch getan hätte. Trotzdem hat es auf Anhieb geklappt und läuft ohne Probleme seit vielleicht 1-2 Jahren.
Soweit ich mich erinnere war für die Installation der Piggy-Backs in den SMA Dokumenten auch gefordert bestimmte Widerstände abzutrennen (je nach SMA Modell). Bei diesem Schritt muss man echt aufpassen denke ich, da habe ich mir als Amateur die größten Sorgen gemacht irgendwas ausversehen kaputt zu machen, weil die Wechselrichter doch etwas schwer zugänglich sind.

Natürlich habe ich vorher mit dem Elektriker der die Anlage installiert hat gesprochen was ich beachten muss, damit das ganze ungefährlich bleibt. Letztendlich habe ich die Sicherungen der Wechselrichter ausgeschaltet und die Verkabelung zu den Solarmodulen getrennt und eine Weile gewartet damit sich Kondensatoren entladen können und was weiß ich.

Erdung und Kabelschlauch für das Datenkabel habe ich bis heute immer noch nicht gemacht, das sollte ich echt mal nachholen. Habt ihr das gemacht bzw. vor?

pkwagner, 2021-03-05:
Danke für die Details.

Zu der letzten Frage kann ich gar nicht viel sagen, die Anlage steht im Haus meiner Eltern - die haben das damals alles von einem Elektriker einbauen lassen. Wegen Covid war ich allerdings schon länger nicht mehr vor Ort und kann deshalb nicht mehr sagen, als dass drei Kabel für RX/TX/GND aus einem der beiden WR ausgeführt sind 😄

chrishrb, 2021-03-15:
also für alle, die auch vorhaben, den adapter zu kaufen, den ich zuerst gekauft habe: macht es NICHT. Ich habe Ewigkeiten rumprobiert, es wollte einfach nicht funktionieren - und habe dann doch noch einen USB-Adapter (den von pkwagner) gekauft. 
Einstecken, starten und hat sofort funktioniert! Danke nochmals, an euch - ihr habt mir sehr geholfen.

closed 2021-03-15 by chrishrb

pkwagner, 2021-03-15:
Sehr schön, freut mich!

Aus reiner Neugier: Wie schaut der Postprocessing-Stack eigentlich bei Dir aus? Ich hatte das für mich mal grob hier[1] skizziert (TL;DR: `InfluxDB` und `Grafana`) und bin auch ziemlich zufrieden, aber natürlich immer offen für bessere Ideen 🌝 

[1] https://github.com/pkwagner/yasdi2mqtt/issues/2#issuecomment-624300271