pkwagner/yasdi2mqtt #11: MQTT client doesn'r reconnect

chrishrb, 2021-03-25:
Hello, 
I don't know if you know this issue, but my client (with TLS) doesn't reconnect after the connection failed one time (e.g. no internet, mqtt server is down, etc.). I have to restart the program, after that it runs normally. 
I don't know if this is a problem with my mqtt server or it's a client error? 

These messages are shown after the reconnect: 
```
yasdi2mqtt_1  | 09:14:30 WARN  src/mqtt_client.c:146: Lost connection to mqtt broker: (null)
yasdi2mqtt_1  | 09:14:30 WARN  src/mqtt_client.c:105: Unknown error while connecting to mqtt broker (status: -8). Going to retry in 10 seconds...
yasdi2mqtt_1  | 09:14:40 WARN  src/mqtt_client.c:105: Unknown error while connecting to mqtt broker (status: -8). Going to retry in 10 seconds...
yasdi2mqtt_1  | 09:14:41 ERROR src/mqtt_client.c:137: Unable to publish via mqtt (status: -3)
yasdi2mqtt_1  | 09:14:41 ERROR src/mqtt_client.c:137: Unable to publish via mqtt (status: -3)
yasdi2mqtt_1  | 09:14:50 WARN  src/mqtt_client.c:105: Unknown error while connecting to mqtt broker (status: -8). Going to retry in 10 seconds...
yasdi2mqtt_1  | 09:15:00 WARN  src/mqtt_client.c:105: Unknown error while connecting to mqtt broker (status: -8). Going to retry in 10 seconds...
yasdi2mqtt_1  | 09:15:10 WARN  src/mqtt_client.c:105: Unknown error while connecting to mqtt broker (status: -8). Going to retry in 10 seconds...
yasdi2mqtt_1  | 09:15:11 ERROR src/mqtt_client.c:137: Unable to publish via mqtt (status: -3)
yasdi2mqtt_1  | 09:15:11 ERROR src/mqtt_client.c:137: Unable to publish via mqtt (status: -3)
```

Thank you in advance.
Greetings Christoph

pkwagner, 2021-03-28:
Sorry for the late answer, I'm currently in the middle of a relocation. I've already experienced a few mqtt disconnects (without TLS), but the reconnection procedure worked out fine for me every time yet. Therefore, I'm pretty sure that this error is related to TLS handshaking, that is maybe not repeated correctly upon reconnect.

I don't have that much free time in the next weeks, so unfortunately, I can't promise a timely fix at the moment. If you want to have a look on your own before - the reconnection magic happens [here](https://github.com/pkwagner/yasdi2mqtt/blob/master/src/mqtt_client.c#L146). `mqtt_connect()` simply calls `MQTTClient_connect` again on the running `MQTTClient`, so shutting down the previous instance and creating a new one instead could be one possible bugfix.

chrishrb, 2021-04-08:
I figured out the problem. If you move this paragraph to the function `mqtt_connect()`, the reconnection does work without problems. I think in the function `MQTTClient_SSLOptions_initializer` there are some timestamps set, which won`t work in the `mqtt_init()` method. I don`t know if this is just a workaround or the final solution. I created a pull request for this issue.

```c
    if (ssl_cert_global != NULL)
    {
        log_debug("Setting up ssl for mqtt connection");

        MQTTClient_SSLOptions ssl_config = MQTTClient_SSLOptions_initializer;
        ssl_config.trustStore = ssl_cert_global;
        ssl_config.sslVersion = MQTT_SSL_VERSION;

        options.ssl = &ssl_config;
    }

```

commit d23fb6b8 (2021-04-17): Fix failing mqtt reconnects when using ssl (fixes #11)
PR #13 (2021-05-16): Fix failing mqtt reconnects when using ssl (@chrishrb 👀)
Fixes #11 as discussed in #12.
commit deb8c207 (2021-05-16): Fix failing mqtt reconnects when using ssl (fixes #11)
closed 2021-05-16 by pkwagner