pkwagner/yasdi2mqtt #28: yasdi2mqtt docker-compose start but no data fro USB RS485

Sn4k3E4t3r, 2022-09-29:
Hi leider funktioniert bei mir der docker-compose weg aktuell nicht. Ich habe einen USB RS485 Stick gekauft. Diesen mit meinem Pi 4 verbunden und hoffentlich alles bei der Anleitung richtig gemacht. 

Nach dem Starten von docker-compose up hängt sich dieser auf. Nach min. 5 Min Warten habe ich dann mit Strg +C den Prozess beendet?

Hat jemand eine Idee was das sein könnte?

USB Stick
https://www.amazon.de/gp/product/B081NBCJRS/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

`root@hsa-iobrokerpi1:/home/pi/yasdi2mqtt# docker-compose up
Starting yasdi2mqtt_yasdi2mqtt_1 ... done
Attaching to yasdi2mqtt_yasdi2mqtt_1
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:38: Configuration | YASDI_CONFIG = /etc/yasdi2mqtt/yasdi.ini
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:39: Configuration | YASDI_DRIVER_ID = 0
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:40: Configuration | YASDI_MAX_DEVICE_COUNT = 1
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:41: Configuration | YASDI_UPDATE_INTERVAL = 30
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:42: Configuration | MQTT_TOPIC_PREFIX = solar/inverter
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:43: Configuration | MQTT_SERVER = 10.11.30.60
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:44: Configuration | MQTT_PORT = 1883
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:45: Configuration | MQTT_SSL_CERT = (null)
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:46: Configuration | MQTT_QOS_LEVEL = 2
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:47: Configuration | MQTT_USER = iobrokermqtt
yasdi2mqtt_1  | 05:46:26 INFO  src/main.c:48: Configuration | MQTT_PASSWORD = ixxxxxxxxxxxxxx
yasdi2mqtt_1  | 05:46:26 INFO  src/mqtt_client.c:89: Connection to mqtt broker established

^CGracefully stopping... (press Ctrl+C again to force)
Stopping yasdi2mqtt_yasdi2mqtt_1 ... done
`

Sn4k3E4t3r, 2022-09-29:
[image]

Ich habe mal LOG_LEVEL: 0 hinzugefügt 

Wird das uahc jetzt im dunkel funktionieren? Oder nur wenn er an ist?

Sn4k3E4t3r, 2022-09-29:
Hi danke dir. Sag mal fährt der den Docker Compose erst komplett hoch wenn der mal ein Device gefunden hat?

pkwagner, 2022-09-29:
Nein, der Container sollte schon komplett oben sein. Du kannst das (über `docker-compose up -d`) also problemlos bis morgen im Hintergrund laufen lassen und dann im Log nachschauen.

Sn4k3E4t3r, 2022-09-30:
Guten Morgen,

es geht und ich hatte A und B verdreht. Trotz das ich eigentlich der DOku gefolgt bin.. :) Danke dir

pkwagner, 2022-09-30:
Sehr gut, freut mich zu hören :)

closed 2022-09-30 by pkwagner

pkwagner, 2022-09-30:
Sorry, die zweite Frage hab ich heute früh überlesen. Schau doch mal hier[1] vorbei, meiner Erinnerung nach hatte @AlexanderSch90 genau das selbe Problem.

[1] https://github.com/pkwagner/yasdi2mqtt/issues/2#issuecomment-626284396