pkwagner/yasdi2mqtt #31: fails to recover from network interruption

randyoo, 2022-11-11:
I'm experiencing a similar issue to #11 at the moment. If my network connection drops, I stop getting MQTT updates, and the logs are filled with:
`
16:34:27 ERROR src/mqtt_client.c:140: Unable to publish via mqtt (status: -3)
16:34:35 ERROR src/mqtt_client.c:140: Unable to publish via mqtt (status: -3)
16:34:47 ERROR src/mqtt_client.c:140: Unable to publish via mqtt (status: -3)
16:34:50 ERROR src/mqtt_client.c:140: Unable to publish via mqtt (status: -3)
16:34:56 ERROR src/mqtt_client.c:140: Unable to publish via mqtt (status: -3)
16:35:01 ERROR src/mqtt_client.c:140: Unable to publish via mqtt (status: -3)
16:35:06 ERROR src/mqtt_client.c:140: Unable to publish via mqtt (status: -3)
`
A restart of the container fixes it, but only until the next network glitch. I'm looking for a way to work around it with a watchdog, but figured I'd report it here, too.

pkwagner, 2022-11-16:
Hi, thanks for reporting! If I recall correctly, the issue you are referring to only affected SSL connections and has been resolved back in the days - but interestingly, I've just experienced the same problem on my own setup.

Therefore, I suspect it has something to do with a recent `paho` (the underlying MQTT library) update. I'll try to have a deeper look in the next days, but for the moment, I would recommend simply stepping back to `stable-alpine` (or `stable`) - at least that's what I did. If my theory holds, the error shouldn't occur there.

randyoo, 2022-11-16:
> <snip>for the moment, I would recommend simply stepping back to `stable-alpine` (or `stable`) - at least that's what I did. If my theory holds, the error shouldn't occur there.

Thanks! Giving it a shot now--I'll report back if it recurs on stable.

closed 2023-10-13 by pkwagner