pkwagner/yasdi2mqtt #34: docker is running, but no Data

frankol, 2023-07-12:
I hope this project is still alive.
the problem is have is that a device is found, but i dont see any data coming.

here the debug messages:
`10:37:29 INFO  src/main.c:65: Configuration | yasdi_config = /config/yasdi.ini
10:37:29 INFO  src/main.c:66: Configuration | yasdi_driver_id = 0
10:37:29 INFO  src/main.c:67: Configuration | yasdi_max_device_count = 1
10:37:29 INFO  src/main.c:68: Configuration | yasdi_update_interval = 30
10:37:29 INFO  src/main.c:69: Configuration | mqtt_topic_prefix = solar/inverter
10:37:29 INFO  src/main.c:70: Configuration | mqtt_server = 192.168.178.120
10:37:29 INFO  src/main.c:71: Configuration | mqtt_port = 1883
10:37:29 INFO  src/main.c:72: Configuration | mqtt_qos_level = 2
10:37:29 INFO  src/main.c:73: Configuration | mqtt_user = otto
10:37:29 INFO  src/main.c:74: Configuration | mqtt_password = otto3361
10:37:29 INFO  src/mqtt_client.c:73: Connection to mqtt broker established
10:37:29 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
10:37:29 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
10:37:29 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
10:37:31 DEBUG src/yasdi_handler.c:247: Event 0 fired for device 1
10:37:31 INFO  src/yasdi_handler.c:271: Device 1 went online
10:37:36 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:36 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:36 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:36 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:36 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:36 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:37 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:37 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:37 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:37 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:37 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:37 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:38 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:38 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
[...]
10:37:53 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:53 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:53 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:53 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:53 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:53 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:54 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:54 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:54 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:54 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:54 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:54 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:54 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:54 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:55 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:55 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:55 DEBUG src/yasdi_handler.c:247: Event 3 fired for device 1
10:37:55 DEBUG src/yasdi_handler.c:295: Downloading channel list for device 1
10:37:55 DEBUG src/yasdi_handler.c:247: Event 2 fired for device 1
10:37:55 DEBUG src/yasdi_handler.c:292: Device detection finished with 1 devices
10:37:59 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
10:37:59 DEBUG src/yasdi_handler.c:158: Device channel list download is apparently still in progress, skipping device 1...
10:37:59 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...`

Any ideas where i can look?

randyoo, 2023-07-12:
Yes, the project is alive, but perhaps seems inactive since it's already completed.

Have you successfully retrieved data from the device *outside* this project? That way you could narrow down the scope of the issue. Could it be a hardware issue (wire termination?), for example?

What is the outcome if you try to connect via SMA's YASDI software in CLI?

It's important that you don't have 2 open YASDI connections at the same time, as they will conflict with and interrupt one another. (ie WebBox or CLI YASDI)

Finally, you can speed things up if you increase the baud rate (Have to change it on the device and in the settings). I doubt it would cure your problem, but couldn't hurt... Viel Glück!

pkwagner, 2023-07-12:
Thanks @randyoo for already handling this!

Another possibility (that seems to be confirmed by your log file [1]) would be that the channel list download is still in progress. Usually this takes around 2-3 minutes and until it's completed, no data can arrive. It's a good sign that yasdi is downloading data at all; your inverter is probably properly connected and set up.

So maybe just try giving it some time and see if after 30 minutes or so data arrives. By the way, if you follow the setup in the README, the device data should be cached and the download hence only be necessary on first startup.

---

Last line says: `Device channel list download is apparently still in progress, skipping device 1...`

frankol, 2023-07-12:
thank you guys, i found out the problem. First of all, im using the docker container in Home Assistant and had to change 2 things. 
1. Add Port Mappings of 2 ports

> ports:
   24273/udp: 24273
   24273/tcp: 24273
   24272/udp: 24272
   24272/tcp: 24272

Without these ports it just dont work and i get the messages above without transmitting anything.

2. change the name of the values field name. Home Assistant has a problem to extract the values out of the json string if the fieldname is "values" :-)

changed:
`cJSON_AddItemToObject(device_json, "values", device_values);`
to
`cJSON_AddItemToObject(device_json, "val", device_values);`

But i would like to know why many values are missing in the json output. I took a look at the downloaded bin file from the inverter and i can see way more values i could get from like

```
Inv.TmpValMax   grdC
Inv.TmpVal          grdC
Pcb.TmpValMax  grdC
Pcb.TmpVal         grdC
BtPwr                  dBm 
```
How can i read these out too?

Is this the limitation? 

DWORD channel_count = GetChannelHandlesEx(device, channels, MAX_CHANNEL_COUNT, **SPOTCHANNELS**);

randyoo, 2023-07-13:
@frankol — Great to hear! I thought about suggesting checking communication with the MQTT server… glad you figured it out. 

I’m afraid I don’t know the answer, but if it’s possible to obtain those temperature values, I’d also be keen to have them, so please let me know if you succeed.

By the way, I had the same problem with “values” being interpreted by Home Assistant as a class (I think), and was able to get around it by using escape characters.

frankol, 2023-07-13:
@randyoo i guess i need still some help. I changed src/yasdi_handler.c but the changes wont get applied. Do you know how i could manage to get my yasdi_handler.c applied?

i also found out how to get more channels, but its the same problem..need my yasdi_handler.c :-)

pkwagner, 2023-10-13:
Sorry, I'm a bit late here - how did you solve your problem? Just by increasing `MAX_CHANNEL_COUNT`?

If so, I would consider changing the variable in this project, so that future users don't run into the same problem (and so that you can use `:latest` again!).

frankol, 2023-10-13:
> Sorry, I'm a bit late here - how did you solve your problem? Just by increasing `MAX_CHANNEL_COUNT`?
Yes, just set it to ALLCHANNELS, but still temperatures are missing

pkwagner, 2023-10-13:
Ah, got it - then I'm going add an additional flag for that.

How many additional channels do you roughly get? Do you think the current `MAX_CHANNEL_COUNT = 300` default is still a viable choice?

closed 2025-04-19 by pkwagner