pkwagner/yasdi2mqtt #41: Docker compose fails

kommando828, 2023-10-06:
I enter the command

docker run \
>    --device /dev/ttyUSB0:/dev/ttyUSB0 \
>    -v <project_dir>/devices:/etc/yasdi2mqtt/devices \
>    -v <project_dir>/yasdi.ini:/etc/yasdi2mqtt/yasdi.ini:ro \
>    -e YASDI_CONFIG="/etc/yasdi2mqtt/yasdi.ini" \
>    -e YASDI_DRIVER_ID="0" \
>    -e YASDI_MAX_DEVICE_COUNT="1" \
>    -e YASDI_UPDATE_INTERVAL="30" \
>    -e MQTT_TOPIC_PREFIX="/solar/inverter" \
>    -e MQTT_SERVER="example.com" \
>    -e MQTT_PORT="1883" \
>    -e MQTT_USER= null\
>    -e MQTT_PASSWORD=null \
>

and get the error

bash: project_dir: No such file or directory

I have created the folder yasdi2mqtt subfolder devices in etc plus created yasdi.ini in etc/yasdi2mqtt as they did not exist but still get the error after a reboot.

Running the Docker pull just confirms that it is up to to date

docker pull pkwagner/yasdi2mqtt
Using default tag: latest
latest: Pulling from pkwagner/yasdi2mqtt
Digest: sha256:00a68b81b47754b7b2733b753da104815d928afaae5387c9f332fe2f19356c38
Status: Image is up to date for pkwagner/yasdi2mqtt:latest

Never used docker before so not doubt a novice error.

pkwagner, 2023-10-06:
Yes, you have to replace `<project_dir>` with your working directory 😉

Also make sure that the `yasdi.ini` configuration is present and that there’s an empty `devices` subdirectory inside `<project_dir>`.

kommando828, 2023-10-07:
Thanks will give it a go later and report back.

kommando828, 2023-10-07:
Ok that worked, now working through editing yasdi.ini to get that sorted, copied across a working Win11 version but the serial port used needs to reflect the PI not a Windows com port.

pkwagner, 2023-10-07:
Oh, that’s a tricky question - I haven’t tried it with Windows yet. Since I don’t provide any Windows container, I guess you’re running the Linux build under some kind of Hyper-V emulation.

And if that’s the case, [it seems like sharing devices with the container is not possible at all](https://learn.microsoft.com/en-us/virtualization/windowscontainers/deploy-containers/hardware-devices-in-containers#run-a-container-with-a-device) (“Device assignment and device sharing for workloads in Hyper-V-isolated Linux containers is not currently supported.”). 

The only solutions I see is either switching to Linux (using a VM?) or rebuilding the whole container for Windows on your own.

kommando828, 2023-10-07:
No its running on a Le Potato which is a close copy of a PI3b, but will shortly be going to a 3b as there are issues I think are related to it being a Le Potato such as Portainer slowing it to a complete crawl on the network so had to uninstall it.

lsusb gives

Bus 001 Device 010: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

Now just have to find the correct form of /dev/ttyUSB0 so it is using that device,

so now need to get its

pkwagner, 2023-10-07:
Ah, got it.

The cheap solution would be to do a simple `ls /dev/tty*` before and after connecting the adapter to see what has changed. Alternatively, you can try `dmesg`, which always logged the ports for me yet.

kommando828, 2023-10-07:
Thumbs up to dmesg

ch341-uart converter now attached to ttyUSB0

so will try that

kommando828, 2023-10-07:
Ok, moving there slowly, can get Yasdishell to open com port by editing the yasdi.ini in the Yasdishell folder where I changed the protocol from RS485 to RS232, ttySO to ttyUSB0 and removed some odd formating which had added extra spaces. Can test the connection for data transfer with the Sunny Island tomorrow but based on the Windows Yasdishell this should work fine. 

However using the same yasdi.ini in etc/yasdi2mqtt results in

04:30:56 FATAL src/yasdi_handler.c:42: Error while initializing yasdiMaster: -1
06:25:16 FATAL src/main.c:58: Unable to initialize yasdi_handler

Which based on previous discussions means the yasdi.ini file needs work.

So after the comms check tomorrow I will compare both yasdi.ini files to make sure they really are exactly the same, easy to do a O instead of a 0. Also check the formatting but that did seem ok when reopening using Nano.

Only other issue is the MQTT broker takes 3 mins before a connection is established.

kommando828, 2023-10-08:
************************************************************

        YASDI Mini Shell (build for Linux)

      (This is an test program using YASDI)

************************************************************
Switching driver 'COM1' on...success
Ok, coms work

Command ('?' for help): e
Starting device detection (sync).
How many devices should be searched? 1
New device found: 'SI5048EI SN:1260013210'

This was a Yasdi standalone install using

wget https://github.com/pkwagner/yasdi/archive/refs/heads/master.zip
unzip master.zip

So its the same yasdi as in yasdi2mqtt, so any issues are down to either my yasdi.ini or my docker settings. 

Also found out this Le Potato does not support Node-red  Pi GIPO so going to start again with a Pi4b.

The plan is to monitor the SI frequency and switch on and off loads using the Pi GIPO's, when the SI sees the batteries nearing full it increases the AC frequency to control the output of SMA inverters. So once the AC frequency increases then by turning on a load the PV output will not be reduced.

dalklein, 2023-10-08:
Kommando, that's the idea that I do with my SI6048 inverters and AC coupled solar to limit curtailing.  Frequency is also available on the can bus at a very fast rate.  From yasdi at 19200 baud, about 3 or 4 second update rate, I use SI battery current and voltage vs target voltage, to control extra loads.  There's a discussion at diysolarforum about rs485 sunny Island communication with yasdi or webbox.

kommando828, 2023-10-08:
Glad to know what I plan can be achieved. I think my SI5048 on the EU firmware or maybe hardware is different from the SI6048US, on my SI is shows 19200 baud rate in the options but I can actually only connect at max of 9600, if I select 19200 there is no comms but Sunny Data Control on my PC is very responsive at 9600 so I will stick to 9600.  Its night and day from the 1200 baud on my Sunny Boys but they are not connected to the SI. I joined diysolar forum recently and there is plenty of good info on there.  Webbox is not an option in UK at £500 despite being obsolete when a Rasp Pi is £40. Node-red also give the option of JSON to modbus TCP so I can data log on my Solar-log which only supports the SI8.0H.  I have already fooled it into detecting one by creating a Modbus Slave with the registers the solar-log searches for in the detection phase, once detected it then only reads the registers that update such as SOC and voltage which again can be taken from JSON. Likewise I use a Janitza UMG96S2 for a meter when the Solar-log only supports the UMG104, created a slave modbus for a UMG104 for detection phase then once detected swapped the rs485 cable to an S96-S2 and it logs the meter readings as if it was a UMG104. Luckily other than the serial number and device address registers most of the Janitza meters updated registers for power, volts, current are all the same.

kommando828, 2023-10-08:
Ok, pi4b is setup, yasdi.ini seems to be correct as its a copy of the one I used in Yasdishell earlier. But still getting the Yasdi handler problem. So there must be an issue with the Docker commands I give but its not obvious to me.

So first the Dmesg for the USB rs485.

 4374.772400] usbcore: registered new interface driver ch341
[ 4374.772504] usbserial: USB Serial support registered for ch341-uart
[ 4374.772635] ch341 1-1.3:1.0: ch341-uart converter detected
[ 4374.776495] usb 1-1.3: ch341-uart converter now attached to ttyUSB0

Which confirms ttyUSB0

The etc/yasdi2mqtt folder is made, etc/yasdi2mqtt/devices folder is made and in etc/yasdi2mqtt is yasdi.ini which reads as

[DriverModules]
Driver0=yasdi_drv_serial

[COM1]
Device=/dev/ttyUSB0
Media=RS485
Baudrate=9600
Protocol=SMANet

[Misc]
#DebugOutput=/dev/stderr

and this is the commands

docker run \
>    --device /dev/ttyUSB0 \
>    -v /etc/yasdi2mqtt/devices \
>    -v /etc/yasdi2mqtt/yasdi.ini \
>    -e YASDI_CONFIG="/etc/yasdi2mqtt/yasdi.ini" \
>    -e YASDI_DRIVER_ID="0" \
>    -e YASDI_MAX_DEVICE_COUNT="1" \
>    -e YASDI_UPDATE_INTERVAL="30" \
>    -e MQTT_TOPIC_PREFIX="/solar/inverter" \
>    -e MQTT_SERVER="192.168.1.172" \
>    -e MQTT_PORT="1883" \
>    -e MQTT_USER=null \
>    -e MQTT_PASSWORD=null \
>    pkwagner/yasdi2mqtt:latest

to which I get

02:29:20 INFO  src/main.c:38: Configuration | YASDI_CONFIG = /etc/yasdi2mqtt/yasdi.ini
01:40:16 INFO  src/main.c:39: Configuration | YASDI_DRIVER_ID = 0
01:40:16 INFO  src/main.c:40: Configuration | YASDI_MAX_DEVICE_COUNT = 1
01:40:16 INFO  src/main.c:41: Configuration | YASDI_UPDATE_INTERVAL = 30
01:40:16 INFO  src/main.c:42: Configuration | MQTT_TOPIC_PREFIX = /solar/inverter
01:40:16 INFO  src/main.c:43: Configuration | MQTT_SERVER = 192.168.1.172
01:40:16 INFO  src/main.c:44: Configuration | MQTT_PORT = 1883
01:40:16 INFO  src/main.c:45: Configuration | MQTT_SSL_CERT = (null)
01:40:16 INFO  src/main.c:46: Configuration | MQTT_QOS_LEVEL = 2
01:40:16 INFO  src/main.c:47: Configuration | MQTT_USER = null
01:40:16 INFO  src/main.c:48: Configuration | MQTT_PASSWORD = null
23:09:08 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
23:06:40 INFO  src/mqtt_client.c:89: Connection to mqtt broker established
12:56:32 FATAL src/yasdi_handler.c:42: Error while initializing yasdiMaster: -1
04:02:20 FATAL src/main.c:58: Unable to initialize yasdi_handler

I will try again tomorrow, any tips would be appreciated.

kommando828, 2023-10-09:
Ok, I assumed the <project_dir> was the /etc/yasdi2mqtt but reading up on Docker its inside the Docker folders /var/lib/docker plus then looking inside the container.

pkwagner, 2023-10-09:
* `<project_dir>` can point to any directory, as long as it contains a `yasdi.ini` configuration and a `devices` directory.
* The errors in your log suggest that something’s wrong with your MQTT connection. At first glance, `MQTT_USER=null` and `MQTT_PASSWORD=null` seem wrong. If you don’t have username/password set, try removing both arguments entirely. I would suppose that you are currently setting both to string value `null`.

kommando828, 2023-10-09:
MQTT is set to not need a user or password so deleted the 2 lines but the problem remains.

 docker run \
>    --device /dev/ttyUSB0:/dev/ttyUSB0 \
>    -v /etc/yasdi2mqtt/devices \
>    -v /etc/yasdi2mqtt/yasdi.ini \
>    -e YASDI_CONFIG="/etc/yasdi2mqtt/yasdi.ini" \
>    -e YASDI_DRIVER_ID="0" \
>    -e YASDI_MAX_DEVICE_COUNT="1" \
>    -e YASDI_UPDATE_INTERVAL="30" \
>    -e MQTT_TOPIC_PREFIX="/solar/inverter" \
>    -e MQTT_SERVER="192.168.1.172" \
>    -e MQTT_PORT="1883" \
>    pkwagner/yasdi2mqtt:latest
04:52:16 INFO  src/main.c:38: Configuration | YASDI_CONFIG = /etc/yasdi2mqtt/yasdi.ini
04:03:12 INFO  src/main.c:39: Configuration | YASDI_DRIVER_ID = 0
04:03:12 INFO  src/main.c:40: Configuration | YASDI_MAX_DEVICE_COUNT = 1
04:03:12 INFO  src/main.c:41: Configuration | YASDI_UPDATE_INTERVAL = 30
04:03:12 INFO  src/main.c:42: Configuration | MQTT_TOPIC_PREFIX = /solar/inverter
04:03:12 INFO  src/main.c:43: Configuration | MQTT_SERVER = 192.168.1.172
04:03:12 INFO  src/main.c:44: Configuration | MQTT_PORT = 1883
04:03:12 INFO  src/main.c:45: Configuration | MQTT_SSL_CERT = (null)
04:03:12 INFO  src/main.c:46: Configuration | MQTT_QOS_LEVEL = 2
04:03:12 INFO  src/main.c:47: Configuration | MQTT_USER = (null)
04:03:12 INFO  src/main.c:48: Configuration | MQTT_PASSWORD = (null)
12:22:44 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 WARN  src/mqtt_client.c:108: Unknown error while connecting to mqtt broker (status: -1). Going to retry in 10 seconds...
12:20:16 INFO  src/mqtt_client.c:89: Connection to mqtt broker established
15:19:28 FATAL src/yasdi_handler.c:42: Error while initializing yasdiMaster: -1
17:15:56 FATAL src/main.c:58: Unable to initialize yasdi_handler

Other than installing MQTT I have just set it to not need a user or password and to be set as service on boot. I have double checked and etc/yasdi2mqtt/devices is present and yasdi2mqtt folder has yasdi.ini.

I then ran it with sudo

 sudo docker run \
>    --device /dev/ttyUSB0:/dev/ttyUSB0 \
>    -v /etc/yasdi2mqtt/devices \
>    -v /etc/yasdi2mqtt/yasdi.ini \
>    -e YASDI_CONFIG="/etc/yasdi2mqtt/yasdi.ini" \
>    -e YASDI_DRIVER_ID="0" \
>    -e YASDI_MAX_DEVICE_COUNT="1" \
>    -e YASDI_UPDATE_INTERVAL="30" \
>    -e MQTT_TOPIC_PREFIX="/solar/inverter" \
>    -e MQTT_SERVER="192.168.1.172" \
>    -e MQTT_PORT="1883" \
>    pkwagner/yasdi2mqtt:latest
04:28:48 INFO  src/main.c:38: Configuration | YASDI_CONFIG = /etc/yasdi2mqtt/yasdi.ini
03:39:44 INFO  src/main.c:39: Configuration | YASDI_DRIVER_ID = 0
03:39:44 INFO  src/main.c:40: Configuration | YASDI_MAX_DEVICE_COUNT = 1
03:39:44 INFO  src/main.c:41: Configuration | YASDI_UPDATE_INTERVAL = 30
03:39:44 INFO  src/main.c:42: Configuration | MQTT_TOPIC_PREFIX = /solar/inverter
03:39:44 INFO  src/main.c:43: Configuration | MQTT_SERVER = 192.168.1.172
03:39:44 INFO  src/main.c:44: Configuration | MQTT_PORT = 1883
03:39:44 INFO  src/main.c:45: Configuration | MQTT_SSL_CERT = (null)
03:39:44 INFO  src/main.c:46: Configuration | MQTT_QOS_LEVEL = 2
03:39:44 INFO  src/main.c:47: Configuration | MQTT_USER = (null)
03:39:44 INFO  src/main.c:48: Configuration | MQTT_PASSWORD = (null)
18:10:28 INFO  src/mqtt_client.c:89: Connection to mqtt broker established
14:56:00 FATAL src/yasdi_handler.c:42: Error while initializing yasdiMaster: -1
23:03:40 FATAL src/main.c:58: Unable to initialize yasdi_handler

and the MQTT errors are gone but the yasdi_handler issue remains.

kommando828, 2023-10-09:
Left working on a docker start as nothing seems to clear the yasdi_handler issues and swapped to 'docker compose up' after putting a docker-compose.yml file in etc/yasdi2mqtt folder. Verbose logging shows the search for devices was started but this was with no connection to the SI just the USB RS485 plugged in. So its now back outside with the SI and running but nothing showing yet with the MQTT issues continuing. But no Yasdi_handler faults so letting it do its thing until it finds the SI.

kommando828, 2023-10-09:
docker compose up
[+] Running 1/0
 ✔ Container yasdi2mqtt-yasdi2mqtt-1  Created                                                                                                                                           0.0s
Attaching to yasdi2mqtt-yasdi2mqtt-1
yasdi2mqtt-yasdi2mqtt-1  | 17:29:36 INFO  src/main.c:38: Configuration | YASDI_CONFIG = /etc/yasdi2mqtt/yasdi.ini
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:39: Configuration | YASDI_DRIVER_ID = 0
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:40: Configuration | YASDI_MAX_DEVICE_COUNT = 1
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:41: Configuration | YASDI_UPDATE_INTERVAL = 30
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:42: Configuration | MQTT_TOPIC_PREFIX = solar/inverter
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:43: Configuration | MQTT_SERVER = 192.168.1.172
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:44: Configuration | MQTT_PORT = 1883
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:45: Configuration | MQTT_SSL_CERT = (null)
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:46: Configuration | MQTT_QOS_LEVEL = 2
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:47: Configuration | MQTT_USER = (null)
yasdi2mqtt-yasdi2mqtt-1  | 16:40:32 INFO  src/main.c:48: Configuration | MQTT_PASSWORD = (null)
yasdi2mqtt-yasdi2mqtt-1  | 18:59:32 INFO  src/mqtt_client.c:89: Connection to mq                                                                                                             tt broker established
yasdi2mqtt-yasdi2mqtt-1  | 01:42:40 INFO  src/yasdi_handler.c:271: Device 1 went                                                                                                              online

kommando828, 2023-10-10:
Need to upgrade the network in the SI area, this morning putty is not connecting reliably to the Pi which is possibly the cause on the MQTT disconnects. Need to add an 8 port GB switch and run a cable to the Pi instead of using the wifi and then recommence the testing.

dalklein, 2023-10-10:
👍 I stopped using the rpi wifi also, only ethernet.

kommando828, 2023-10-11:
Update

Network is upgraded, still get the MQTT issue but on the second connection it stays connected. I am taking that as a result for now.

However still would not get past Device 1 went online, left it running for hours and no progress.

So then installed plain Yasdi to see if that worked but after detecting the SI and listing its serial number but it also never progressed further before stopping and saying no devices detected. But Yasdi had worked in the Le Potato, so next I changed RS485 adaptor to an older one in case Raspbian in the Pi 4 has an issue, this is still a CH340 chip but an earlier version and 2 wire instead of 3.

With the new adaptor Yasdi works once, it detects the SI5048 and adds it to the device list, you can then get parameter lists etc. However if you close and then rerun Yasdi it will no longer detect the SI. A reboot of the PI allows Yasdi to work one more time before not working again.

Docker Compose up gives a different result regardless of whether its run directly after a reboot or after Yasdi and also if run several times.

yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 272
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 273
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 274
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 275
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 276
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 277
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 278
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 279
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 280
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 281
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 282
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 283
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 284
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 285
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 286

Changing the Yasdi.ini file from rs232 to rs485 makes no difference.

So it looks like the CH340 chip driver in Pi Raspbian is not running correctly and I need to get another adaptor based on another chip. I will got for an FT232 based adaptor.

Later I will put the Le Potato back in place and see if that works with Docker Compose as I never tried that on the Le Potato. Their Raspbian version is the same Buster release but written for their hardware so maybe they loaded the driver for the CH340 differently.

kommando828, 2023-10-11:
Test with Le Potato

Yasdishell connects to the SI and will list the Channel values and the Parameters but only with the newer RS485 USB adaptor with a CH341G chip. With the older adaptor there is no connection at all.

Got MQTT sorted for now

So went for Docker Compose and get

yasdi2mqtt-yasdi2mqtt-1  | 18:14:28 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
yasdi2mqtt-yasdi2mqtt-1  | 23:56:04 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 186
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 187
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 188
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 189
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 190
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 191
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 192

So waiting on a FT232 chip based USB adaptor

kommando828, 2023-10-13:
New adaptor with FDTI FT232 chip arrived, put it on the Le Potato

Result is

Yasdishell works, recognises SI and loads spot and parameter channels.

Yasdi2mqtt gives the following error messages, scrolls up the screen as per the CH341 and this is just a sample.

yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 186
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 187
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 188
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 189
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 190
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 191
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 192
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 193
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 194
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 195
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 196
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 197
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 198
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 199
yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 200

So now stumped, line 217 of yasdi_handler looks to refer to line 202 which looks at char value_str[MAX_VALUE_STR_SIZE] which is set to 10 on line 11. 

Yasdishell on the Le Potato gives the following on spot channels

Channel handle |    Channel name    | Channel value (Unit) |
------------------------------------------------------------
     186       | '             Msg' | '0.000'
     187       | '             Soh' | '98.000' (%)
     188       | '     TotInvPwrAt' | '-0.100' (kW)
     189       | '       TotInvCur' | '1.700' (A)
     190       | '     TotInvPwrRt' | '-0.300' (kVAr)
     191       | '          BatSoc' | '95.500' (%)
     192       | '          BatVtg' | '50.300' (V)
     193       | '      BatChrgVtg' | '55.971' (V)
     194       | '        AptTmRmg' | '0.000' (hhmmss)
     195       | '       TotBatCur' | '7.600' (A)
     196       | '          BatTmp' | '11.900' (degC)
     197       | '        RmgTmFul' | '29.000' (d)
     198       | '        RmgTmEqu' | '168.000' (d)
     199       | '       BatSocErr' | '4.500' (%)
     200       | '         GnRmgTm' | '0.000' (hhmmss)
     201       | '        InvPwrAt' | '-0.100' (kW)
     202       | '    InvPwrAtSlv1' | '0.000' (kW)
     203       | '    InvPwrAtSlv2' | '0.000' (kW)
     204       | '    InvPwrAtSlv3' | '0.000' (kW)
     205       | '          InvVtg' | '230.000' (V)
     206       | '      InvVtgSlv1' | '0.000' (V)
     207       | '      InvVtgSlv2' | '0.000' (V)
     208       | '      InvVtgSlv3' | '0.000' (V)
     209       | '          InvCur' | '1.700' (A)
     210       | '      InvCurSlv1' | '0.000' (A)
     211       | '      InvCurSlv2' | '0.000' (A)
     212       | '      InvCurSlv3' | '0.000' (A)
     213       | '          InvFrq' | '49.000' (Hz)
     214       | '        InvPwrRt' | '-0.300' (kVAr)
     215       | '    InvPwrRtSlv1' | '0.000' (kVAr)
     216       | '    InvPwrRtSlv2' | '0.000' (kVAr)
     217       | '    InvPwrRtSlv3' | '0.000' (kVAr)
     218       | '        ExtPwrAt' | '0.000' (kW)
     219       | '    ExtPwrAtSlv1' | '0.000' (kW)
     220       | '    ExtPwrAtSlv2' | '0.000' (kW)
     221       | '    ExtPwrAtSlv3' | '0.000' (kW)
     222       | '          ExtVtg' | '4.300' (V)
     223       | '      ExtVtgSlv1' | '0.000' (V)
     224       | '      ExtVtgSlv2' | '0.000' (V)
     225       | '      ExtVtgSlv3' | '0.000' (V)
[... 97 lines cut ...]
     327       | '           Error' | '-----'
     328       | '          ChpStt' | 'Idle'
     329       | '      PvGdConStt' | '---'
     330       | '     LodGdConStt' | '---'
     331       | '       BatMntStt' | 'Off'

You can see

296       | '   Serial Number' | '1260013210.000'

where the SN is 14 digits long as somehow the SN has a decimal point and 000 added.

On Yasdishell windows 1.8.1 build9 I also get

Device handle: 1
Device 'SI5048EI SN:1260013210' has 142 Spot channels:
Reading channel values, please wait...

 296       | '   Serial Number' | '1260013210.000'

I am using firmware  7.210/7.200 which is the last European release for the SI5048.

So are these extra .000 causing the error on line 217 as the limit of 10 digits being breeched ?

Can this be increased to 14 without affecting other installs?

pkwagner, 2023-10-13:
Yes, that’s very possible. Could you try increasing the limit to let’s say 50?

All you have to do is to increase the variable in line 11 and compile the container on your own using `docker build --no-cache -t my_own_y2m` inside the main workdir. You may then run the image using the name `my_own_y2m `.

kommando828, 2023-10-13:
Ok, dumb questions time as Docker is new to me and very confusing.

As this is docker which runs in a container is the working directory etc/yasdi2mqtt or Docker folders /var/lib/docker plus then looking inside the container. 

acc19c066656   pkwagner/yasdi2mqtt:latest   "yasdi2mqtt"   2 hours ago   Up 12 minutes                       yasdi2mqtt-yasdi2mqtt-1

Or do I copy the full git and put in a new folder, edit the yasdi_handler file and then run docker build --no-cache -t my_own_y2m in the new folder.

kommando828, 2023-10-13:
Reading discussions the SN with the .000 is not unique

https://github.com/pkwagner/yasdi2mqtt/discussions/18#discussioncomment-2172453

394       | '   Serial Number' | '2100521179.000' 

but this did not stop yasdi2mqtt working.

https://github.com/pkwagner/yasdi2mqtt/discussions/18#discussioncomment-2179207

Message 8 received on solar/SB4000TL20/2100521179 at 09:09:

{
    "sn": 2100521179,
    "time": 1644912539,
    "values": {
        "A.Ms.Amp": 0.4930000234162435,
        "A.Ms.Vol": 363.0699918847531,
        "A.Ms.Watt": 178,
        "B.Ms.Amp": 0.41200001956894994,
        "B.Ms.Vol": 181.9899959322065,
        "B.Ms.Watt": 74,
        "Pac": 244,
        "GridMs.Hz": 49.99999888241291,
        "GridMs.PhV.phsA": 233.4999947808683,
        "GridMs.PhV.phsB": 0,
        "GridMs.PhV.phsC": 0,
        "E-Total": 41766.055983781116,
        "Mt.TotOpTmh": 42438.93208324094,
        "Mt.TotTmh": 43603.99742670983,
        "Op.TmsRmg": 0,
        "Serial Number": 2100521179,
        "Op.EvtCntUsr": 3982,
        "Op.EvtNo": 0,
        "Mode": "MPP",
        "Error": "-------",
        "Op.GriSwStt": "Cls",
        "Inv.TmpLimStt": "None",
        "MainModel": "Solar-WR"
    }
}

QoS: 0 - Retain: false

kommando828, 2023-10-13:
Went back to Yasdi2mqtt version 1.2 and its working, Yah.

"sn": 1260013210,
  "time": 1697208728,
  "values": {
    "Msg": 0,
    "Soh": 98,
    "TotInvPwrAt": -0.5999023541808128,
    "TotInvCur": 3.1000977009534836,
    "TotInvPwrRt": -0.09990234673023224,
    "BatSoc": 98.60009912401438,
    "BatVtg": 54.000000804662704,
    "BatChrgVtg": 55.835004494115765,
    "AptTmRmg": 0,
    "TotBatCur": -12.79990253597498,
    "BatTmp": 13.300097852945328,
    "RmgTmFul": 29,
    "RmgTmEqu": 167,
    "BatSocErr": 6.100000090897083,
    "GnRmgTm": 0,
    "InvPwrAt": -0.5999023541808128,
    "InvPwrAtSlv1": 0.00009765475988388062,
    "InvPwrAtSlv2": 0.00009765475988388062,
    "InvPwrAtSlv3": 0.00009765475988388062,
    "InvVtg": 230.00000342726707,
    "InvVtgSlv1": 0,
    "InvVtgSlv2": 0,
    "InvVtgSlv3": 0,
    "InvCur": 3.1000977009534836,
    "InvCurSlv1": 0.00009765475988388062,
    "InvCurSlv2": 0.00009765475988388062,
    "InvCurSlv3": 0.00009765475988388062,
    "InvFrq": 48.90000072866678,
    "InvPwrRt": -0.19990234822034836,
    "InvPwrRtSlv1": 0.00009765475988388062,
    "InvPwrRtSlv2": 0.00009765475988388062,
    "InvPwrRtSlv3": 0.00009765475988388062,
    "ExtPwrAt": 0.00009765475988388062,
    "ExtPwrAtSlv1": 0.00009765475988388062,
    "ExtPwrAtSlv2": 0.00009765475988388062,
    "ExtPwrAtSlv3": 0.00009765475988388062,
    "ExtVtg": 4.400000065565109,
    "ExtVtgSlv1": 0,
    "ExtVtgSlv2": 0,
    "ExtVtgSlv3": 0,
    "ExtCur": 0.30009765923023224,
    "ExtCurSlv1": 0.00009765475988388062,
    "ExtCurSlv2": 0.00009765475988388062,
    "ExtCurSlv3": 0.00009765475988388062,
    "ExtFrq": 0,
    "ExtPwrRt": 0.00009765475988388062,
    "ExtPwrRtSlv1": 0.00009765475988388062,
    "ExtPwrRtSlv2": 0.00009765475988388062,
    "ExtPwrRtSlv3": 0.00009765475988388062,
    "TotExtPwrAt": 0.00009765475988388062,
    "TotExtCur": 0.30009765923023224,
    "TotExtPwrRt": 0.00009765475988388062,
    "TotLodPwr": -0.5999023541808128,
    "GdRmgTm": 0,
    "Pac": -0.5999023541808128,
    "Firmware": 7.210000342456624,
    "Iac": 3.1000977009534836,
    "Vac": 229.90000342577696,
    "Fac": 48.90000072866678,
    "ChpPwrAt": 0.00009765475988388062,
    "ChpRmgTm": 0,
    "ChpStrRmgTm": 0,
    "Sic1PvPwr": 0,
    "Sic2PvPwr": 0,
    "Sic3PvPwr": 0,
    "Sic4PvPwr": 0,
    "TotSicPvPwr": 0,
    "TotSicBatCur": 0.00009765475988388062,
    "TotMccLodPwr": 0.00009765475988388062,
    "TotPvPwrAt": 0,
    "TotLodPwrAt": 0,
    "SlfCsmpPwrAt": 0,
    "SlfCsmpIncPwr": 0,
    "BatCpyThrpCnt": 1,
    "GdCsmpPwrAt": 0,
    "GdFeedPwrAt": 0,
    "PacPV": 0.00009765475988388062,
    "PacFeed-In": 0.00009765475988388062,
    "PacConsumption": 0.00009765475988388062,
    "kWhPV": 0,
    "FwVer": 7.210000342456624,
    "OnTmh": 88568.90131977946,
    "FwVer2": 7.200000341981649,
    "EgyCntIn": 43430.60064716637,
    "EgyCntOut": 28757.300428517163,
    "EgyCntTm": 87931.60131028295,
    "GnEgyCnt": 44960.70066996664,
    "GnEgyTm": 19812.500295229256,
    "GnOpTmh": 19881.400296255946,
    "GnStrCnt": 56438,
    "GdEgyCntIn": 0,
    "GdEgyCntOut": 108.10000161081553,
    "GdEgyTmh": 0,
    "GdOpTmh": 2266.0000337660313,
    "GdCtcCnt": 0,
    "TotTmh": 0,
    "Sic1EgyCntIn": 0,
    "Sic2EgyCntIn": 0,
    "Sic3EgyCntIn": 0,
    "Sic4EgyCntIn": 0,
    "TotSicEgyCntIn": 0,
    "Sic1TdyEgyCntIn": 0,
    "Sic2TdyEgyCntIn": 0,
    "Sic3TdyEgyCntIn": 0,
    "Sic4TdyEgyCntIn": 0,
    "TotSicDyEgyCntI": 0,
    "Serial Number": 1260013210,
    "E-Total-In": 43430.60064716637,
    "E-Total": 28757.300428517163,
    "h-On": 88568.90131977946,
    "TotLodEgyCnt": 0,
    "SlfCsmpIncEgy": 0,
    "SlfCsmpEgy": 0,
    "SlfCsmpIncTdy": 0,
    "GdCsmpEgyTdy": 0,
    "kWhFeed-In": 0,
    "kWhConsumption": 0,
    "GdFeedEgyTdy": 0,
    "Adr": "Master",
    "OpStt": "Operating",
    "OpSttSlv1": "---",
    "OpSttSlv2": "---",
    "OpSttSlv3": "---",
    "CardStt": 2,
    "Prio": "Ina",
    "BatChrgOp": "Float",
    "AptPhs": "Off",
    "GnDmdSrc": "None",
    "GnStt": "Off",
    "InvOpStt": "Run",
    "InvOpSttSlv1": "---",
    "InvOpSttSlv2": "---",
    "InvOpSttSlv3": "---",
    "Rly1Stt": "Off",
    "Rly2Stt": "Off",
    "GnRnStt": "Off",
    "Mode": "Operation",
    "Error": "-----",
    "ChpStt": "Idle",
    "PvGdConStt": "---",
    "LodGdConStt": "---",
    "BatMntStt": "Off"
  }
}

pkwagner, 2023-10-13:
Hm, that's strange. MQTT problems with newer releases are not uncommon, as the build pipeline always pulls the latest `paho` (the MQTT library) source code - but the underlying SMA YASDI library for device communication didn't change in decades. So I didn't expect any difficulties there.

Did you maybe use `:alpine` before? `:alpine` and `:1.2-alpine` have a different build target than `:latest` and `:1.2`.

pkwagner, 2023-10-13:
Okay, thanks for reporting. I think it's a good start that at least the stable version works 😄 

Can I close this issue?

kommando828, 2023-10-13:
Let me add the results from tomorrows tests and then it can be closed. Might as well see if 1.3 shows as working so if someone has the same issue then 1.3 can be a solution two. Looking at the container ID's 1.3 and stable look to be from the same build as the 6 container ID's are identical.

pkwagner, 2023-10-13:
Yes, `:stable` and `:1.3` are currently identical.

The idea is that `:stable` always points to the last manually tested release (`:1.3` in this case), while `:latest` and `:alpine` are automatically built (without any testing) by the pipeline every month to include potential security fixes in the underlying MQTT library.

kommando828, 2023-10-13:
I will have cloned the SD card before any test anyway, plus the containers are separate too. If 1.2 is the last version that works for a Buster OS Pi would you be able to do an alpine version of 1.2 ?

pkwagner, 2023-10-13:
Unfortunately not, all releases are automatically compiled using my GitHub actions pipeline, so I don't have it set up on my local machine.

However, the source code for all releases is archived [here](https://github.com/pkwagner/yasdi2mqtt/releases). You can just unzip the folder on your target device and compile your own image using the mentioned `docker build --no-cache -t my_own_y2m`.

For the Alpine build, you'll have to copy over the `Docker.alpine` from the current branch to the old version and run `docker build --no-cache -f Dockerfile.alpine -t my_own_alpine_y2m`.

kommando828, 2023-10-14:
Got the 1.2 zip file for later once I get some SD cards I am waiting on so I can start playing.

This morning the 1.2 was still running outputting messages with Sunny Data Control in tandem.

Cloned SD card to an image as a backup in case something goes wrong later.

Set the Pi 4b going again and changed to Ver 1.1

docker compose up
[+] Running 1/0
 ✔ Container yasdi2mqtt-yasdi2mqtt-1  Created                              0.0s
Attaching to yasdi2mqtt-yasdi2mqtt-1
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:65: Configuration | yasdi_c                                                                                                             onfig = /etc/yasdi2mqtt/yasdi.ini
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:66: Configuration | yasdi_d                                                                                                             river_id = 0
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:67: Configuration | yasdi_m                                                                                                             ax_device_count = 1
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:68: Configuration | yasdi_u                                                                                                             pdate_interval = 30
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:69: Configuration | mqtt_to                                                                                                             pic_prefix = solar/inverter
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:70: Configuration | mqtt_se                                                                                                             rver = 192.168.1.172
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:71: Configuration | mqtt_po                                                                                                             rt = 1883
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:72: Configuration | mqtt_qo                                                                                                             s_level = 0
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:73: Configuration | mqtt_us                                                                                                             er = (null)
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/main.c:74: Configuration | mqtt_pa                                                                                                             ssword = (null)
yasdi2mqtt-yasdi2mqtt-1  | 15:20:47 INFO  src/mqtt_client.c:73: Connection to mq                                                                                                             tt broker established
yasdi2mqtt-yasdi2mqtt-1  | 15:20:48 INFO  src/yasdi_handler.c:271: Device 1 went                                                                                                              online

followed by Mosquitto Explorer showing messages being received 2 or 3 mins later so ver 1.1 works for my setup.

Then changed to Ver 1.3 and got the yasdi_handler 217 error continuous stream

yasdi2mqtt-yasdi2mqtt-1  | 00:00:00 ERROR src/yasdi_handler.c:217: Got invalid value for device 1 / channel 186

Also checked the source code for 1.2 yasdi_handler and it has the same 10 MAX_VALUE_STR_SIZE 10 which is not surprising as it was last changed on Github 3 years ago.

Went back to Ver 1.2 and its working again.

kommando828, 2023-10-14:
Le Potato also works with 1.1 and 1.2 but only after getting 

error getting credentials - err: exit status 1, out: `GDBus.Error:org.freedeskto p.DBus.Error.ServiceUnknown: The name org.freedesktop.secrets was not provided b y any .service files`

which was fixed with

sudo apt install gnupg2 pass

Its a credentials issue.

closed 2023-10-14 by kommando828