pkwagner/yasdi2mqtt #44: Home Assistant not connecting with SMA inverters

shoei600, 2023-11-24:
Hi all,
I'm hoping somebody can help me get over this final stage get things working. I have tried the original setup in a docker on a Windows machine. On startup the script would run fine and connect to mqtt broker, then no further output at all. It was getting too complicated for me so I opted to use the Home Assistant version )https://github.com/frankol/yasdi2mqtt/tree/master) which has the port detail resolved. The site has 5 inverters all with Webconnect modules. Again, the log shows it runs fine, connects to mqtt broker then no further output at all. 
I have full connection to all inverters with Sunny Explorer running on the same windows machine. Home Assistant is running in virtualbox with bridged connection, not had any issues reaching any network device. I have setup each inverter to enable UDP server and changed the port to match the library (24273). I have checked this too with PrtQry which reports the inverters open and responding.
I have copied yasdi.ini to config folder and added IP details:

[DriverModules]
Driver0=yasdi_drv_ip
Driver1=yasdi_drv_serial
[COM1]
Device=/dev/ttyUSB0
Media=RS485
Baudrate=1200
Protocol=SMANet
[IP1]
Protocol=SMANet
Device0=192.168.100.51
[Misc]
DebugOutput=/dev/stdout

The HomeAssistant system logs have nothing other than the below which may help, but I'm at a loss where to go from here. Thank you in advance.

File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 124, in async_unload_entry
return await component.async_unload_entry(entry)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in async_unload_entry
await platform.async_reset()
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 766, in async_reset
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1038, in async_remove
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity update.yasdi2mqtt_update async_remove called twice

shoei600, 2023-11-24:
I seem to be making progress but just this last hurdle to overcome. I moved to a different fork which was clearly setup for IP comms (https://github.com/creuter27/yasdi2mqtt/tree/master) and has HA instructions, but I'm stuck on ow to do this last point "_Integration works if you give the Docker the following command: -p 24273:24273/udp \ => yasdi basically creates a UDP server which waits for input on the port_". How can I tell Home Assistant to do this?

My log is shown below which shows constantly looking for devices but not finding any and I have added devices to the yasdi.ini which I presume is not necessary. 

Should the IP address in the yasdi.ini be for the host machine or SMA devices?
[DriverModules]
Driver0=yasdi_drv_ip
Driver1=yasdi_drv_serial
[COM1]
Device=/dev/ttyUSB0
Media=RS485
Baudrate=1200
Protocol=SMANet
[IP1]
Protocol=SMANet
Device0=192.168.100.51
[Misc]
DebugOutput=/dev/stdout

0:19:02 INFO  src/main.c:44: Configuration | YASDI_CONFIG = /config/yasdi.ini
20:19:02 INFO  src/main.c:45: Configuration | YASDI_DRIVER_ID = 0
20:19:02 INFO  src/main.c:46: Configuration | YASDI_MAX_DEVICE_COUNT = 5
20:19:02 INFO  src/main.c:47: Configuration | YASDI_UPDATE_INTERVAL = 30
20:19:02 INFO  src/main.c:48: Configuration | MQTT_TOPIC_PREFIX = solar/inverter
20:19:02 INFO  src/main.c:49: Configuration | MQTT_SERVER = 192.168.100.48
20:19:02 INFO  src/main.c:50: Configuration | MQTT_PORT = 1883
20:19:02 INFO  src/main.c:51: Configuration | MQTT_SSL_CERT = (null)
20:19:02 INFO  src/main.c:52: Configuration | MQTT_CLIENT_ID = yasdi2mqtt
20:19:02 INFO  src/main.c:53: Configuration | MQTT_QOS_LEVEL = 2
20:19:02 INFO  src/main.c:54: Configuration | MQTT_USER = mqttuser
20:19:02 INFO  src/main.c:55: Configuration | MQTT_PASSWORD = xxx
20:19:03 INFO  src/mqtt_client.c:88: Connection to mqtt broker established
20:19:03 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:19:03 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:19:03 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
20:19:33 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:19:33 DEBUG src/yasdi_handler.c:108: Device detection already running, won't start new query
20:19:33 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:19:33 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
20:20:03 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:20:03 DEBUG src/yasdi_handler.c:108: Device detection already running, won't start new query
20:20:03 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:20:03 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
20:20:33 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:20:33 DEBUG src/yasdi_handler.c:108: Device detection already running, won't start new query
20:20:33 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:20:33 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
20:21:04 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:21:04 DEBUG src/yasdi_handler.c:108: Device detection already running, won't start new query
20:21:04 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:21:04 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
20:21:34 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:21:34 DEBUG src/yasdi_handler.c:108: Device detection already running, won't start new query
20:21:34 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:21:34 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
20:22:05 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:22:05 DEBUG src/yasdi_handler.c:108: Device detection already running, won't start new query
20:22:05 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:22:05 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
20:22:36 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:22:36 DEBUG src/yasdi_handler.c:108: Device detection already running, won't start new query
20:22:36 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:22:36 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
20:23:06 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:23:06 DEBUG src/yasdi_handler.c:108: Device detection already running, won't start new query
20:23:06 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:23:06 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...
20:23:37 DEBUG src/yasdi_handler.c:97: Not all devices are online, starting device detection (async)...
20:23:37 DEBUG src/yasdi_handler.c:108: Device detection already running, won't start new query
20:23:37 DEBUG src/yasdi_handler.c:147: Starting device value fetch (sync)...
20:23:37 DEBUG src/yasdi_handler.c:136: yh_loop is going to sleep for 30 seconds...

shoei600, 2024-05-02:
Hi Paul, I never could get this working and was not able to get any support with it.  I now communicate with them via modbus which has its own issues and SMA have now told me they are unable to support modbus even though it's built in to their products. So I would like to revisit it.

pkwagner, 2024-05-05:
#47 might be related to your issue, as TCP does not seem to be enabled in the current version of `yasdi2mqtt`. 

Have you tried running your config with `yasdishell`?

MOS6510, 2024-05-06:
Hi shoei600,

I think you have to put every device into the same '**IP0**' driver section 'cause they collides with the same local listen port (with no further parameter). When you use multiple sections for IP instead then all 'DeviceX' entryies should start with "Device**0**" again.
So I suggest you to try again with this setup:

--- cut ---
[DriverModules]
Driver0=yasdi_drv_ip
Driver1=yasdi_drv_serial

[COM1]
Device=/dev/ttyUSB0
Media=RS485
Baudrate=1200
Protocol=SMANet

[IP0]
Protocol=SMANet
Device0=192.168.100.54
**Device1**=192.168.100.51
**Device2**=192.168.100.52
**Device3**=192.168.100.55
**Device4**=192.168.100.234

[Misc]
DebugOutput=/dev/stdout
--- cut ---

If you communicate with UDP only you can remove the "COM1" section for serial communication. 

It's easy to check if this works. Recompile yasdi and enable debug output. Start 'yasdishell' and it should output something like this where all IP addresses are listed as 'Added new peer...':

[image]

closed 2025-04-19 by pkwagner