pkwagner/yasdi2mqtt #46: Sending data

kr0ner, 2024-04-16:
Hi there, first of all thanks for the great work! I'm running your images within Home Assistant and it works like a charm.

On first startup the device config was downloaded and I just displayed the content:
It contains bunch of stuff like:
```
jGER / ENSGB  / G77AUS / AGLUSA / UL1741Res1Res2Res3Res4Res5Res6Res7Res8Res9Res10Res11Res12Storage        1permanentvolatileRes1Res2Res3Res4Res5Res6       Upv-Ist        V      ?     Upv-Soll       V      ?     Iac-Ist        mA     ?     "Iac-Soll       %      ?    Uac            V      ?     Fac            Hz
#<     Pac            W             Zac            Ohm    o:            "dZac           Ohm    o:
       RErd-Start     kOhm   ?     "Uac-Srr        V      ?    "Fac-Srr        Hz
<      "Zac-Srr        Ohm    o:  "Izac           A      o:  "Tkk            grdC   =    Ipv            mA     ?     "Tkk max        grdC   =    "Upv max        V      ?    "Uzwk           V      ?    dI             mA     ?     "dI-Srr         mA     ?    E-Total        kWh    o:   h-Total        h      9     h-On           h      9     Netz-Ein              ?     Fehler-Cnt            ?    Seriennummer          ?      Status  
```

Which seems to match the content of the payload of the MQTT topic 👍 

But there is more:
```
SMA-SN                OUpv-Start      V      BDT-Start        s      @CT-Stop         s      ?CdU U-Konst     V/s    ?AdU-Mpp Track   V/TaMp? AdUref- Mpp     V      @I Such neu     mA     F Usoll-Konst    V      BD
"Td-Ureg        s      ?"Pmax           W      @EPlimit         W      `E`ENiTest                ?"Kom.-Ni               zC"Cnt-Imax              ?zD"Upv-Max        V      BDUac-Min        V      4CCUac-Max        V      4CCFac-delta-     Hz     @Fac-delta+     Hz     @dFac-Max       Hz/s
ף;@dZac-Max       mOhm   D"Riso-Min       kOhm   @Fkk-Max        grdC   B"Upv-Fakt              <A"Iso-Fakt              D;
#<"It-Fakt               C"Iac-Fakt              =B Hardware-BFS   VersionB!Software-BFR   VersionB"Software-SRR   VersionB#"Uac-BFR-Fakt          Ǻ>pΨ>$"Uac-SRR-Fakt          Z??%"AD0-3KorrFakt         G&"AD0-3KorrOffs         @F'"AD4-7KorrFakt         G("AD4-7KorrOffs         @F)Inst.-Code 
```

Have you ever tried writing data? If I remember correctly it is also possible to e.g. reset the counters or set a limit for power output

I'm using an ancient yellow SMA SunnyBoy 2100TL

pkwagner, 2024-04-17:
Thanks for the kind words.

Yes, being able to write to channels would be a nice addition. [We actually had the same discussion two years ago](https://github.com/pkwagner/yasdi2mqtt/discussions/24), and came to the conclusion that such functionality was probably supported by the underlying `yasdi` library (which your `devices` dump seems to confirm). However, to the best of my knowledge, no one has gone ahead with an implementation; and speaking for myself, until I have a use case, I'm not very tempted to play around with the configuration of my inverters.

So unfortunately it's not on my personal roadmap. But if you are interested in doing the work, [I would be very happy to merge it](https://github.com/pkwagner/yasdi2mqtt/discussions/24#discussioncomment-2867475)!

kr0ner, 2024-05-02:
In the end I used the yasdi shell ;)
But I did not manage to set PLimit, cause it is hardcoded ^^ also written somewhere in the official docu :D 
So for me there is no need to do this directly from HA, as those settings are likely to be written once.

closed 2024-05-02 by kr0ner