pkwagner/yasdi2mqtt #47: HASS plugin and RS485 over IP

stich86, 2024-05-04:
Dear @pkwagner,

i'm tryting to run yasdi2mqtt on HASS.IO using you latest image and using a RS485 to WiFi adapter, here is the yasdi.ini config:

`[DriverModules]
Driver0=./libyasdi_drv_ip.so

# Configs for communiation over Ethernet/UDP
# Replace 192.168.178.9 with the real IP address of your device

[IP0]
Protocol=SMANet
Device0=10.0.1.166`

when container starts, it throws this error:

```
22:00:26 INFO  src/main.c:43: Configuration | YASDI_CONFIG = /config/yasdi.ini
22:00:26 INFO  src/main.c:44: Configuration | YASDI_MAX_DEVICE_COUNT = 1
22:00:26 INFO  src/main.c:45: Configuration | YASDI_UPDATE_INTERVAL = 30
22:00:26 INFO  src/main.c:46: Configuration | MQTT_TOPIC_PREFIX = sma/inverter
22:00:26 INFO  src/main.c:47: Configuration | MQTT_SERVER = localhost
22:00:26 INFO  src/main.c:48: Configuration | MQTT_PORT = 1883
22:00:26 INFO  src/main.c:49: Configuration | MQTT_SSL_CERT = (null)
22:00:26 INFO  src/main.c:50: Configuration | MQTT_CLIENT_ID = yasdi2mqtt
22:00:26 INFO  src/main.c:51: Configuration | MQTT_QOS_LEVEL = 2
22:00:26 INFO  src/main.c:52: Configuration | MQTT_USER = hass
22:00:26 INFO  src/main.c:53: Configuration | MQTT_PASSWORD = XXXXX
22:00:27 INFO  src/mqtt_client.c:88: Connection to mqtt broker established
22:00:27 FATAL src/yasdi_handler.c:42: Error while initializing yasdiMaster: -1
22:00:27 FATAL src/main.c:63: Unable to initialize yasdi_handler
```

doing a tcpdump on the hass.io, I don't see any attemps to connect over tcp. The `yasdi.ini` is placed into `/config` folder where Homeassistant config are presents (i've found the bind using docker inspect on the container)

Any ideas?

Thanks in advance

stich86, 2024-05-04:
ok my fault, correct `yasdi.ini` is this one:

```
[DriverModules]
Driver0=yasdi_drv_ip

# Configs for communiation over Ethernet/UDP
# Replace 192.168.178.9 with the real IP address of your device

[IP0]
Protocol=SMANet
Device0=10.0.1.166
```

just for future reference, connection to remote side over Wi-Fi is **UDP** on port  **24272**, but i've seen that yasdi lib should support also TCP

pkwagner, 2024-05-04:
So with the new config, it works for you?

Generally speaking, I have no control over the underlying `yasdi` library, which manages the config and handles the connection to the inverters. If `yasdi` terminates with an exception, my general advice is to first check the config with `yasdishell` (the showcase utility of `yasdi`) first, which is usually much more verbose.

stich86, 2024-05-04:
It's currently starting. I'll test it on next Saturday after install module on my inverter :) 

Do you know how to compile with TCP support? I see that it's off by default (like UDP)

pkwagner, 2024-05-04:
Oh, you're right! TCP is indeed supported by `yasdi` and just [disabled by default](https://github.com/pkwagner/yasdi/blob/1e845bddc2ece8dbf4c4ed7c56dae18a378ca27b/projects/generic-cmake/CMakeLists.txt#L22).

I'll have a look if there's any downside of enabling this flag. In the meantime, you can also try to compile a modified container on your own by adding `-D YASDI_DRIVER_TCP=1` behind the other CMake flag [here](https://github.com/pkwagner/yasdi2mqtt/blob/0783f3d7b5f43b899c09bab887891b7eff1aba2c/Dockerfile.alpine#L18C52-L18C75) and running `docker build -f Dockerfile.alpine -t my_own_yasdi2mqtt .`.

stich86, 2024-05-04:
I'll try to test both udp and tcp next week and report back.

If they works l, I'll create a PR with are needed changes ;)

pkwagner, 2024-05-04:
Great, thanks!

stich86, 2024-05-05:
so.. looks like that TCP driver will not build:

```
-- Building YASDI Version '1.8.1Build9' (Copyright by SMA Solar Technology AG)
CMake Error at CMakeLists.txt:146 (INCLUDE):
  INCLUDE could not find load file:

    tcp_driver.cmake
```

the file `tcp_driver.cmake` (and also bluetooth one) are not present in the yasdi lib repository.. may be is not present in the original build from SMA?

MOS6510, 2024-05-06:
Hi! The TCP driver (and also the bluetooth driver) are not included in the original SMA software package because they had the state of "experimetal" only. No SMA inverter device supported this directely at that time.

stich86, 2024-05-06:
> Hi! The TCP driver (and also the bluetooth driver) are not included in the original SMA software package because they had the state of "experimetal" only. No SMA inverter device supported this directely at that time.

uh.. perfect :)
ok i'll test it directly on UDP ;)

thanks for the feedback!

stich86, 2024-05-11:
Hi @pkwagner

i've installed adapter to my inverter and connected yasdi2mqtt over UDP to my RS485 adapter, looks like that the app sees an online device, but nothing more:

```
13:05:08 INFO  src/main.c:43: Configuration | YASDI_CONFIG = /config/yasdi.ini
13:05:08 INFO  src/main.c:44: Configuration | YASDI_MAX_DEVICE_COUNT = 1
13:05:08 INFO  src/main.c:45: Configuration | YASDI_UPDATE_INTERVAL = 30
13:05:08 INFO  src/main.c:46: Configuration | MQTT_TOPIC_PREFIX = sma/inverter
13:05:08 INFO  src/main.c:47: Configuration | MQTT_SERVER = localhost
13:05:08 INFO  src/main.c:48: Configuration | MQTT_PORT = 1883
13:05:08 INFO  src/main.c:49: Configuration | MQTT_SSL_CERT = (null)
13:05:08 INFO  src/main.c:50: Configuration | MQTT_CLIENT_ID = yasdi2mqtt
13:05:08 INFO  src/main.c:51: Configuration | MQTT_QOS_LEVEL = 2
13:05:08 INFO  src/main.c:52: Configuration | MQTT_USER = xxxxx
13:05:08 INFO  src/main.c:53: Configuration | MQTT_PASSWORD = xxxxxx
13:05:09 INFO  src/mqtt_client.c:88: Connection to mqtt broker established
13:05:09 INFO  src/yasdi_handler.c:69: Driver IP0 (ID: 0) online
13:05:10 INFO  src/yasdi_handler.c:286: Device 1 went online 
```

Doing a tcpdump I see that there is a response from the adapter, please find attach the pcap

192.168.179.132 is Home Assistant
192.168.179.143 is the RS485 Adapter

[dump_rs485.pcap.zip](https://github.com/pkwagner/yasdi2mqtt/files/15282579/dump_rs485.pcap.zip)

Any help is appreciated

stich86, 2024-05-11:
ok it tooks a little bit, but now it's working :)

[image]

I'll keep it monitored for the next days, then i'll try to create a full PR with also some example and configuration needed for RS485 over IP ;)

pkwagner, 2024-05-11:
Great, glad to hear that!

closed 2024-05-11 by pkwagner